microsoft/pxt-calliope #31: Cannot export staticpkg from the freshly installed calliope target

baradaran, 2018-09-24:
Right after complete installation, i try to export a static version. I face the following error. 
```
INTERNAL ERROR: TypeError: Cannot read property 'replace' of undefined
at Object.renderMarkdown (~/pxt-calliope/node_modules/pxt-core/built/pxt.js:58949:33)
   at renderDocs (~/pxt-calliope/node_modules/pxt-core/built/pxt.js:81652:33)
   at ~/pxt-calliope/node_modules/pxt-core/built/pxt.js:83305:34
   at tryCatcher (~/pxt-calliope/node_modules/bluebird/js/release/util.js:16:23)
   at Promise._settlePromiseFromHandler (~/pxt-calliope/node_modules/bluebird/js/release/promise.js:512:31)
   at Promise._settlePromise (~/pxt-calliope/node_modules/bluebird/js/release/promise.js:569:18)
   at Promise._settlePromise0 (~/pxt-calliope/node_modules/bluebird/js/release/promise.js:614:10)
   at Promise._settlePromises (/~/pxt-calliope/node_modules/bluebird/js/release/promise.js:694:18)
   at Promise._fulfill (~/pxt-calliope/node_modules/bluebird/js/release/promise.js:638:18)
   at Promise._resolveCallback (~/pxt-calliope/node_modules/bluebird/js/release/promise.js:432:57)
   at Promise._settlePromiseFromHandler (~/pxt-calliope/node_modules/bluebird/js/release/promise.js:524:17)
   at Promise._settlePromise (~/pxt-calliope/node_modules/bluebird/js/release/promise.js:569:18)
   at Promise._settlePromise0 (~/pxt-calliope/node_modules/bluebird/js/release/promise.js:614:10)
   at Promise._settlePromises (~/pxt-calliope/node_modules/bluebird/js/release/promise.js:694:18)
   at Promise._fulfill (~/pxt-calliope/node_modules/bluebird/js/release/promise.js:638:18)
   at MappingPromiseArray.PromiseArray._resolve (~/pxt-calliope/node_modules/bluebird/js/release/promise_array.js:126:19)

```
appearantly this has been an issue before [https://github.com/Microsoft/pxt/issues/3612](https://github.com/Microsoft/pxt/issues/3612), but i did not get any info on how was is solved in microbit.

pelikhan, 2018-09-24:
It is not solved for microbit. Calliope is still on pxt v0 for which staticpkg is not functional. Calliope needs to be upgraded to v1 in order to make this work.

baradaran, 2018-09-24:
is there anyway that i can manually solve this?

pelikhan, 2018-09-24:
Afraid not. Upgrade calliope to pxt v4 is needed or fix bugs in v0. None of those are trivial.

baradaran, 2018-09-26:
thanks, is there a high level way point? like a documentation you could point me to? or high level idea of which modules need updates?

Amerlander, 2019-12-02:
staticpkg is working now in the master branch, but there are still some issues which needs to be fixed before a release: https://github.com/microsoft/pxt-calliope/pull/54

closed 2019-12-02 by Amerlander