microsoft/pxt-calliope #44: Pins mapped wrong in Simulator

MKleinSB, 2019-04-14:
The Pins P0, P1, P2, P3 should be marked when high and the analog/digital values should be displayed at the correct place. ( The other pins on the Expansion-port should be marked, too when they are used)
[image]

MKleinSB, 2020-01-19:
@Amerlander Motorpins should be marked too when using the motorblocks and perhaps rotating wheels beneath the calliope like openroberta does?

Amerlander, 2020-01-21:
the 6th motorpin is now also included in the PR https://github.com/microsoft/pxt-calliope/pull/72.
The leftover things will follow these days.

Amerlander, 2020-01-22:
The Pins P0 to P3 are now fixed. The positions are corrected and they are displaying the analog as well as the digital values.
I also added the black circle in the middle of the little pins to the hovering are, so it is a bit easier to hover them with the mouse.

Amerlander, 2020-01-22:
Since we will replace the whole simulator illustration with the actual layout I didn't put too much effort into the visual aspect yet.

MKleinSB, 2020-01-23:
Works great now! Only the unused labels look a little bit funny ;-)
[image]

closed 2020-01-23 by MKleinSB

Amerlander, 2020-01-23:
The unused labels will get removed again, they were just for debugging there