microsoft/pxt-calliope #47: github.io page is not the latest build

baradaran, 2019-05-07:
I have already asked about staticpkg problem  which is still an open issue here : [microsoft/pxt-calliope#31]

As a workaround which helps me for now, there is already a working version hosted here : 
https://makecode.calliope.cc/ 

1 - Can i have access to these hosted files? It will really solve some of my problems. 
2- Where can i access and see how this [static version](https://makecode.calliope.cc) is created given that the pxt V0 statikpkg does not work on the current build. Is it from a specific version or commit ?

Thank you for your time

baradaran, 2019-05-21:
@pelikhan  It would be great if you could answer this.

pelikhan, 2019-05-21:
staticpkg is not supported on the pxt version calliope is built against. It needs to be upgraded.

magicmarvman, 2019-11-08:
@pelikhan @joernalraun 

Are there any plans or schedules to upgrade the version? This would be great, also for me it would be a nice feature...

Apart from that, this repo hasn't been updated for a year.... Are people still working on this MakeCode®?

Amerlander, 2019-11-29:
@magicmarvman we just started a pull request for a new version. It will still be beta, but most things work well.

So work on this MakeCode wasn't that much the last time, but it will get more now.

Amerlander, 2019-12-02:
fixed with https://github.com/microsoft/pxt-calliope/pull/54

closed 2019-12-02 by Amerlander