microsoft/pxt-calliope #57: Error in math.ts when loading extension with .cpp - file

MKleinSB, 2019-12-11:
When loading Extensions with a  .cpp - file, 
(like https://github.com/ferenc-hechler/pxt-remember-int)
an error in math.ts appears
Fehler: pxt_modules/core/math.ts(1,1): error TS9043: The hex file is not available, please connect to internet and try again.
The simulator doesn´t load and the program can´t be compiled.

Amerlander, 2019-12-11:
I'm investigating this issue at the moment, but doesnt have much progress.

It works when the editor is opened on localhost. So running the dev version with `pxt serve` or the staticpkg on a local server (accessed over 172.0.0.1 or localhost) is working, but accessing the same server over the public IP leads to an error in math.ts.

MKleinSB, 2019-12-21:
Perhaps it helps: the math.ts error appears too if you delete radio.

Amerlander, 2020-01-15:
fixed with https://github.com/microsoft/pxt-calliope/issues/67

MKleinSB, 2020-06-04:
Sorry, Error reappears in latest Betaversion. e.g https://github.com/MKleinSB/pxt-kitronik-zip-tile-calliope

Amerlander, 2020-06-04:
I think with upgrading to the latest pxt version this might be solved. I have issues with the recreation of the dal.d.ts file, when they are solved I will push an update, and we can check again.

Amerlander, 2020-09-11:
should work now in the latest beta

closed 2020-09-11 by Amerlander