microsoft/pxt-calliope #64: built of extensions fail

MKleinSB, 2020-01-01:
If you create an extension and create a release it ends in an error because of missing pxt-calliopemini

[image]

Amerlander, 2020-01-03:
This repo needs to be deployed on npmjs as "pxt-calliopemini".

@pelikhan is this something you manage, or should calliope do this on their own?

pelikhan, 2020-01-14:
why not just ``pxt-calliope``?

pelikhan, 2020-01-15:
Latest package available at https://www.npmjs.com/package/pxt-calliope

MKleinSB, 2020-01-17:
@Amerlander nmp is still trying to install pxt-calliopemini instead of pxt-calliope

Amerlander, 2020-01-17:
As far as I see the npm is generating the name out of the target id.

Maybe it makes sense to change the target id to just `calliope` instead of `calliopemini`, since it would be more consistent. But I do not know what consequences will follow when the id gets changed... the build process on the microsoft servers and the crowdin translations are attached to the id.

@pelikhan @joernalraun what do you think?

pelikhan, 2020-01-17:
Can you change your scripts instead?

Amerlander, 2020-01-20:
It is not an issue in the target scripts. I think the only thing we could change without changing pxt itself is changing the target id or the name of the npmjs hosting.

The pxt script creates the `pxt target calliopemini` command out of the target id: `pxt target @TARGET@` (https://github.com/microsoft/pxt/blob/2162d160d11bc700e495a51da3477e70a5c27f2c/pxtlib/initprj.ts#L131)
To work it has to be `pxt target calliope` at the moment.

If we like to change pxt itself I think we need another optional value in the target config like `npmjs-name` or something like that which is used instead of the id if filled out.

Amerlander, 2020-01-20:
To answer your question 'why not just `pxt-calliope`?': I have no idea why the id was named `calliopemini` instead of just `calliope`. But whether we name it `calliope` or `calliopemini` I think it would be an effort to match the names everywhere (target id, github, npmjs, …). And if we should choose to change the id its better done sooner than later.

pelikhan, 2020-01-22:
This is a legacy issue.

The ``calliope`` id points to the editor ``mini.pxt.io``. We needed to create a new id ``calliopemini`` to support the new editor at ``makecode.calliope.cc``.

Amerlander, 2020-09-15:
Should work now, the target is available at `pxt-calliopemini` on npm

closed 2020-09-15 by Amerlander

bexmann, 2022-02-23:
Hi @Amerlander ,
luckily I've found this post. After 2 years of solving the issue I cannot understand that nobody again has this problem but only me today again. Nevermind... I started to use the GitHub integration in MakeCode and get  (after committing a project) the same error code as above. I also cannot import the Calliope Project from GitHub to MakeCode again (nothing is displayed in MakeCode). I hope if you or somebody else can explain what is going wrong and help me! 
[image]

Amerlander, 2022-02-24:
Hi @bexmann,
the error message is the same, but its probably not related to that issue.

When you push through GitHub via the MakeCode site, you receive a notification/email about "Run failed: MakeCode" / "MakeCode: All jobs have failed", correct?
This error is annoying, but you still should be able to use the GitHub integration without other issues. So the import from GitHub to MakeCode should also work. The error messages should disappear, as soon as we are able to ship the new version of MakeCode (https://github.com/microsoft/pxt-calliope/pull/147)

If you still have issues, could you open a new issue and provide the exact steps to reproduce your issue or link to a public repository, that fails to import?

bexmann, 2022-02-24:
Hi @Amerlander,
as @Robo-Lehrer already mentioned through Slack I was able to resolve the issue of importing to MakeCode by setting the repo to public. Thanks for quick reply. Unfortunately I‘m not member of Andreas‘ Slack-Channel to follow further conversation between both of you…

Amerlander, 2022-02-24:
@bexmann, it seems to be a combination of having a private repository and using extensions while trying to initialize the GitHub repository in MakeCode. So removing the extension before creating the repository was also working for me. I was able to add the extension after that again.

Since it is working in https://makecode.microbit.org, which uses a newer version of PXT, I think this issue will be solved when the Calliope target is updated.

bexmann, 2022-02-25:
Hi @Amerlander,
thanks for checking the issue. I did some tests today and came to a slightly different result:
The problem does not occur when using extensions. This always happens when you want to access your private repro with MakeCode. You have initially set the repro to public to be able to import it into MakeCode. After successful import you can then set it to private and the all works as expected as long as you don't remove it from MakeCode again.
So this might be an unexpected behavior of the GitHub integration and we could probably report this somewhere in GitHub to Microsoft, if this isn't already known...

Cheers und schönes Wochenende,
Michael

bexmann, 2022-02-25:
> Hi @Amerlander, thanks for checking the issue. I did some tests today and came to a slightly different result: The problem does not occur when using extensions. This always happens when you want to access your private repo with MakeCode. You have initially set the repo to public to be able to import it into MakeCode. After successful import you can then set it to private and the all works as expected as long as you don't remove it from MakeCode again. So this might be an unexpected behavior of the GitHub integration and we could probably report this somewhere in GitHub to Microsoft, if this isn't already known...
> 
> Cheers und schönes Wochenende, Michael

Maybe @pelikhan can have a look and reply?