microsoft/pxt-calliope #66: Leading characters in German Strings not capitalized correct

MKleinSB, 2020-01-09:
I thought it was a hosting-problem of makecodebeta.calliope.cc,  but then I found the bug in the other makecode editors. Somewhere in the latest makecode or pxt-updates the capitalization was killed:

https://github.com/microsoft/pxt-arcade/issues/1691

Amerlander, 2020-01-09:
yes, I think its something which has to be changed in PXT itself.

The behavior seems for me a bit unpredictable:

Original string is not lower cased:
[image]

Translated string is lower cased, regardless of crowdin translation:
[image]

Most strings are not lower cased:
[image]

At MicroBit there is also no lower casing:
[image]

MKleinSB, 2020-01-09:
The lower casing is the same in https://makecode.microbit.org/beta which has the same pxt version like Calliope mini

[image]

pelikhan, 2020-01-10:
I will look into this

Amerlander, 2020-01-14:
fixed in pxt: https://github.com/microsoft/pxt/pull/6444

closed 2020-01-14 by Amerlander