microsoft/pxt-calliope #70: Missing pictures in tutorial, first steps ...

MKleinSB, 2020-01-17:
The cards in First staps, Tutorial ... stay blank since the new hosting of calliope beta! 
[image]

pelikhan, 2020-01-17:
You have “/docs/“ added to all your images.

Amerlander, 2020-01-18:
Fixed them now.

On our test server the doc images wasn't rendered without this addition, but since local-http server also doesn't need that it has to be a wrong configuration on that server.

closed 2020-01-18 by Amerlander

MKleinSB, 2020-01-23:
@Amerlander : Pictures in Tutorials work, but not those when klicking the Cards!
[image]