microsoft/pxt-calliope #73: Missing and updating extensions

MKleinSB, 2020-01-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Add screenshots to help explain your problem. You can copy paste the screenshot in the github report. The .gif screen recording is very useful as well. 

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

Amerlander, 2020-09-11:
Servo extension is added, groove will be updated, but that has to be done in the extensions repo.

The  RGB-Colorsensor and BC95 Extensions doesnt seem to work and I think if they do not get an update they'll stay out. But you can ping @joernalraun if you think they are important and discuss this with him.

closed 2020-09-11 by Amerlander

Amerlander, 2020-09-11:
Grove is updated now

MKleinSB, 2020-09-11:
BC95 works here:
https://github.com/MKleinSB/pxt-bc95

MKleinSB, 2020-09-11:
sorry, does not work, but is not necessary