microsoft/pxt-calliope #74: add loudness for the microphone in Simulator and Input

MKleinSB, 2020-01-21:
Circuit playground has a microphone, too. You can change the value for loudness in the simulator like the lightlevel or temperature. So it would be nice to have this for Calliope Mini!

[image]

And the way to read the values of the microphone is a little bit difficult/inkonsequent for children. You know Calliope Mini is designed for Children at the age of 9 years. So it would be better to add a loudness Block in Input instead of reading an analog Pin and search the Button for the microphone.

[image]

Thanks! Michael

MKleinSB, 2020-02-22:
This is the needed Code in DAL. Loudness will be from 0 to 100
https://forum.calliope.cc/t/mikrofon-empfindlichkeit/842/13

Amerlander, 2020-02-22:
Thank you, haven't had a look yet, but the enhancing the simulator will be the next thing I do.

If you are working on the code, you can feel free to make changes in this repo (or the one which is hosted in github.org/calliope-edu) and create pull requests.

MKleinSB, 2020-02-23:
Hm, this:
https://github.com/OpenRoberta/microbit-dal/blob/e986dbcae07cf0d76a7ad67050890e1a5a723485/source/drivers/MicroBitPin.cpp#L298
is missing in **your** DAL as far as I see. It only needs to be copied and exposed in basic.cpp to be used as a block. But I´m not a programmer you know ;-)

Amerlander, 2020-02-23:
Me neither, I've studied something different ;)

There is one already working on the dal, so I ask him first if he already integrates it. If not I will change this.

MKleinSB, 2020-02-23:
If not, use my translated code https://github.com/MKleinSB/pxt-mikrofon . Perhaps the easy way.

Amerlander, 2021-11-25:
Targeted in https://github.com/microsoft/pxt-calliope/pull/147

MKleinSB, 2021-12-07:
Hm, I cannot find the three standard functions onLoudSound(), soundLevel() and setLoudSoundThreshold() in your latest beta?

closed 2022-03-22 by Amerlander