microsoft/pxt-calliope #77: "more" submenu in blocks menu not longer working (in beta)

MarkusPiotrowski, 2020-02-04:
makecode.calliope.cc Version:  2.1.45
Microsoft MakeCode Version:  5.32.21
Laufzeitumgebung calliope Version:  pxtgc-0

**Describe the bug**
The submenu "...more" (...mehr) in the blocks menu is not working any longer. The presented blocks stay the same. So you cannot access the additional blocks.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://makecode.calliope.cc/beta#editor
2. Click on e.g. 'Eingabe' (input blocks are shown)
3. Click on '...mehr' (blocks don't change)

**Expected behavior**
Clicking on a '...more' submenu should show the additional blocks.

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**
 - OS: Win7 Prof.
 - Browser: Firefox
 - Version: 72.0.2 (64 bit)

MKleinSB, 2020-02-04:
https://github.com/microsoft/pxt-microbit/issues/2594

MarkusPiotrowski, 2020-02-04:
"Doppelt hält besser" ;-) 

Sorry, I didn't see that one. Always hard to know the correct repository...

closed 2020-02-04 by MarkusPiotrowski

MKleinSB, 2020-02-04:
Du warst schon richtig, aber beim microbit schauen mehr Microsoftmitarbeiter vorbei ;)

pelikhan, 2020-02-04:
Fixed now.