microsoft/pxt-calliope #79: The Mic readings are random. Possibly a hardware bug?

baradaran, 2020-02-07:
**Describe the bug**
In the new hardware version (Version 1.3). The MIC values seem random. There is no accurate read. 

**To Reproduce**
Steps to reproduce the behavior:

Simply screen the pins.analogReadPin(AnalogPin.MIC) . 
The numbers vary between 480 and 530

**Expected behavior**
It should reflect the noise around. Simply attach a previous hardware version and see

**Additional context**
I am not sure if it is something that can be done through software changes. is it simply a production bug?

joernalraun, 2020-02-07:
Well, actually it seems to be implemented as a raw sinus wave and therefore you would need to setup a way to work with the data over a certain time.

baradaran, 2020-02-07:
@joernalraun  I am familiar with the Pulse width modulation concept. However this seems like a normal analogue read to me, plus, an older hardware, calliope ver 1.2 does not have the same issue. I can start with digging deeper with the issue as well, i just thought it is easy to see. 

by the way is there a board schematic from vers 1.3 online?

joernalraun, 2020-02-07:
> @joernalraun I am familiar with the Pulse width modulation concept. 
Yes, but I would say that the value is about an audio wave, thus sinus and not PWM.
You can find the data sheets at calliope-mini.github.io.

baradaran, 2020-02-07:
It does not look like an audio wave, plus the calliope ver1.2 output is just a line in noise free enviroment. so version 1.2 does not act the same. it is only in ver 1.3 which i see this spikes. 

[image]

I cannot make sense of it and would love to read more. please point me to some documentation if you found time.

joernalraun, 2020-02-07:
> It does not look like an audio wave, plus the calliope ver1.2 output is just a line in noise free enviroment. so version 1.2 does not act the same. it is only in ver 1.3 which i see this spikes.

I guess both Calliope minis have the same program running?
From what I remember there has been only software updates for the mic between 1.2 and 1.3 and those should be available to both versions. I‘ll have to dig a bit to find those changes.

MKleinSB, 2020-02-08:
Perhaps this can help you? Sorry, it´s german.
https://www.elektronik-labor.de/Microbit/Calliope2.html

MKleinSB, 2020-02-08:
and the datasheet
https://www.mouser.com/datasheet/2/218/SPU0410HR5H-PB_revH-876884.pdf

joernalraun, 2020-02-08:
> Perhaps this can help you? Sorry, it´s german.
> https://www.elektronik-labor.de/Microbit/Calliope2.html

Please note:
This is for versions below and up until 1.0!

baradaran, 2020-02-12:
> Perhaps this can help you? Sorry, it´s german.
> https://www.elektronik-labor.de/Microbit/Calliope2.html

Thank you for the reply Michael. From what I see, he also only added an amplifier and read the measurements afterward. From his code, I understand that he only reads the difference between two consecutive reads and reports if it is over a threshold. 

This also confirms that there should be no spikes in a silent environment. I still strongly believe its a hardware bug.

by the way, I am impressed with all the sources you have. Maybe start you own 
https://github.com/philipphenkel/awesome-calliope list? :)

joernalraun, 2020-02-12:
@baradaran Your report was not quite clear as you are generally referring to series 1.3 when you where starting this thread and not your device. Of course you can have a hardware bug on this very device. However, please contact support@calliope for this issue as this is not specific about pxt-calliope/MakeCode. I'll close this as this might confuse other users.

closed 2020-02-12 by joernalraun