microsoft/pxt-calliope #93: percent sign missing in "motors"

MKleinSB, 2020-04-01:
There should be a "%" sign at the end of the motorstrings.

[image]

joernalraun, 2020-04-01:
This should be done within Crowdin – I think. I added the percentage symbol there.

MKleinSB, 2020-04-01:
Thanks, but the english strings must be changed, too.

MKleinSB, 2020-04-01:
Hm, you can add % but it never appears. Perhaps the wrapper eats it. He thinks it´s a Variable.

joernalraun, 2020-04-01:
Maybe, I copied from another string. I could see that you added it somewhere a while ago. However, I put it in a different field where it wasn't included.

MKleinSB, 2020-04-17:
I found the solution. I had to use backslash \%

MKleinSB, 2020-04-18:
Ah, should be added in the english source strings, too.

closed 2020-04-21 by pelikhan