microsoft/pxt-calliope #103: Wrong font in repository type

MKleinSB, 2020-05-08:
The font to select the repository type differes from the rest of the gui.

[image]

Amerlander, 2020-05-08:
What browser are you using?

Just testet in Chrome and there it looks fine:
[image]

And how does it look in the microbit editor?

MKleinSB, 2020-05-08:
It´s Firefox and the microbiteditor looks correct.

Amerlander, 2020-05-08:
Nope, its nor your cache. I can confirm this for Firefox.

Amerlander, 2020-05-08:
Firefox has dropped the support for custom fonts in selection-dropdowns. It seems like only system-fonts are applied. At the microbit editor the font in option-dropdowns is also not the same as they are using on the rest of the site., but they have a better fallback defined.

I'll add Courier and sans-serif as fallback, so it looks more similar.

Amerlander, 2020-05-09:
closed by https://github.com/microsoft/pxt-calliope/pull/104

closed 2020-05-09 by Amerlander