microsoft/pxt-calliope #113: No sound on Calliope

MKleinSB, 2020-09-09:
There is no sound when saving the hex including music blocks to the mini.
And when using selectpitchpin there should be the possibility to select the buzzer.
Using play melody shows an analog output at P1. Using set value the output changes.
Using start melody shows an output of 0.
Something wrong with the pins?
makecode.calliope.cc version:  3.0.26 
[image]

pelikhan, 2020-09-09:
@Amerlander i think the default output pin is P0 right now.

Amerlander, 2020-09-09:
I just opened a PR for fixing this.

SelectPitchPin will not work at all at the moment. The internal sound of the mini is differently handled, so the pitch functions are  never called.
So some other sound blocks like set volume will not work and I think we will hide them for now and bring the beta to stable and add them back later when we add support for selectPitchPin.

closed 2020-09-11 by Amerlander