microsoft/pxt-calliope #126: No Blocks appear in playground

MKleinSB, 2020-10-08:
No Blocks appear in playgound. 
microbit and other works, bot not Calliope mini.

[image]

joernalraun, 2020-10-08:
When I check the default one (hello) it does work for me. Have you tried different code?

Amerlander, 2020-10-08:
It also works for me.
Except the WordPicker seems a bit strange, since it picks letters not words and even this doesn't work - but that's also happening in the other editors.

Do you get any errors in the browsers console? (CTRL+Shift+I und dann den Reiter "Konsole" wählen)
I get some in Firefox, but it still works. No errors are appearing in Chrome.

[image]

[image]

MKleinSB, 2020-10-09:
Tested it today again, now it works!
Perhaps only a problem with cache...
Sorry!

closed 2020-10-09 by MKleinSB