microsoft/pxt-calliope #128: Search in approved extensions doesn´t work well

MKleinSB, 2020-11-10:
Searching in extensions doesn´t work as good as in makecode for microbit. It should be enough do type ds18b20 or colorbit but:

[image]

With the correct path it works. But who knows the path for every extension?

[image]

Well, it should work like the search on micro:bit.

joernalraun, 2020-11-10:
Extensions have to be entered manually if you want them easily accessible. If you have good extensions that make sense to be included for the Calliope mini, just send them to us and we will take care of this (with the next release).

MKleinSB, 2020-11-11:
Yes thanks, but the colorbit for example **is** in approved repos and so should appear in search!

[image]

joernalraun, 2020-11-12:
Oh, you are right. I didn't see it there at first.

Amerlander, 2022-03-23:
Should be fixed in the next release

closed 2022-03-23 by Amerlander