microsoft/pxt-calliope #132: ".hex" not compatible to Micro:Bit v2

chatelao, 2021-02-15:
**Describe the bug**
The ".hex" is not compatible to Micro:Bit v2

**To Reproduce**
Steps to reproduce the behavior:
1. Create a Sketch
2. Download to Micro:Bit v2 (not Calliope itself)

**Expected behavior**
Having German samples working for the Micro:Bit v2 is nice.

joernalraun, 2021-02-15:
I think the expected way is to load a Calliope mini project (.hex) to the specific MakeCode version of your choice and build it from there. Hardware specific changes are build directly into the .hex file during the save process. When you save from Calliope mini MakeCode you will always get a Calliope mini specific .hex file.

MKleinSB, 2021-02-15:
Simply switch to Javascriptmode and copy and paset the code.

chatelao, 2021-02-16:
Thank you very much for your hints, they would all work. My problem is another:

- I hoped to find at/for Calliope German tutorials for my local kids
- As a gift for groups the Micro:Bit is much more affordable (15 vs. 40 Euros)

Therefore a firmware compatibility would be very nice.
 But maybe I'm wrong and the tutorials are anyway shared over both projects and translated synchronous?

joernalraun, 2021-02-16:
Ok, I see. That will not be possible. There are simply hardware differences, as the Calliope mini has (for example) an RGB LED and an (amazing) absolute orientation sensor and a motor driver... Something that is not found on another device is hard to take care of in tutorials. Sorry! It would require two completely different made sets of the same examples (and you would not be able to automatically switch). Upcoming tutorials will also require Grove connectors that are also only directly implemented on the Calliope mini.

chatelao, 2021-02-16:
Okay, no problem.

closed 2021-02-16 by chatelao