microsoft/pxt-calliope #137: Shrinking expandable Arguments kills them

MKleinSB, 2021-04-18:
Expandle Arguments don´t work, even in playground. They need to be fully expanded or they generate an error of missing Arguments. 
[image]

switching to Javascript shows that the arguments are missing

[image]

Amerlander, 2021-04-18:
I commented in the micro:bit issue with a solution on this. Are there any expandable blocks in the editor itself which are causing this issue or is it only that sample in the playground?

MKleinSB, 2021-04-19:
Thanks that works now for me!
In the Editor itself there are no expandable Blocks. Should we close this issue?
[image]

Amerlander, 2021-04-19:
I think, Its not an issue in this repo, but a misleading sample in the playground which has to be fixed - but the playground is independent of this repo.

closed 2021-04-19 by Amerlander