microsoft/pxt-calliope #145: setpitchpin still not working

MKleinSB, 2021-09-12:
https://github.com/microsoft/pxt-calliope/issues/113

For the work with children it is much easier to use two crocodileclips on Ground and Pin P1 instead of connecting the motorpin to a speaker. 
When will there be an update to handel the sound of the calliope mini like the microbit does this?

Thx  Michael

tianchrist, 2023-06-03:
I know this is two years old but it may still help people who find this thread.
I found the following code in the docs that works with my Calliope in MakeCode. This code will output the sound on P1 just like with the micro:bit. So you can use crocodile clips. The on-board speaker will not ouput sound, however.

```
pins.setAudioPin(AnalogPin.P1)
pins.analogPitch(440, 1000)
```

MKleinSB, 2023-06-04:
Ja, aber die Musikblöcke sollten auch funktionieren.

closed 2024-03-28 by MKleinSB