microsoft/pxt-calliope #161: Choosing RAM seize does not work

Amerlander, 2022-03-24:
We have implemented a setting to choose between 16 and 32kb Target. But regardless of what switch is on, the hex file is always compiled with the settings for 32kb RAM.

Is it even possible to change the Ram size in that way? Are there better options, like choosing a device every time a new project is created as we see it in https://maker.makecode.com?

[image]

```json
"userConfigs": [
            {
                "description": "32K RAM (mini v2 and some v1.3)",
                "config": {
                    "microbit-dal": {
                        "stack_size": 2048,
                        "sram_end": "0x20008000"
                    },
                    "RAM_SIZE": "\"32K\""
                }
            },
            {
                "description": "16K RAM (mini v1.3 and below)",
                "config": {
                    "microbit-dal": {
                        "stack_size": 1280,
                        "sram_end": "0x20004000"
                    },
                    "RAM_SIZE": "\"16K\""
                }
            }
        ]
```

Amerlander, 2022-03-31:
@pelikhan, can you (or can you name us someone who can) assist us regarding this issue?

pelikhan, 2022-03-31:
This should work since it is the mechanism to enable/disable BLE as well.

I think you need to make sure those RAM_SIZE is mapped to a constant that starts with ``YOTTA_CFG_``. See https://github.com/lancaster-university/microbit-dal/blob/602153e9199c28c08fd2561ce7b43bf64e9e7394/inc/platform/yotta_cfg_mappings.h#L23.

If you add ?dbg=1 or build locally, you should see all the generated build files and in particular the constants that are generated by pxt.

Best to reach out for the MakeCode team.

Amerlander, 2022-03-31:
well, when I add `?dbk=1` the Parameter in `yotta.json` is always `32K` no matter what I choose in the setting.

Also when I build it locally it is always the value which is default in `optionalConfig`. When I change the defaults to `16K` the build process builds the 16k version as expected.

https://makecode.calliope.cc/app/f3b8cb87d8c932e5673206a3808d227e36964416-1e24dc8c20?dbg=1#editor

[image]
[image]

pelikhan, 2022-03-31:
Try using yotta instead of build config

https://makecode.calliope.cc/app/f3b8cb87d8c932e5673206a3808d227e36964416-1e24dc8c20?dbg=1#editor<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fmakecode.calliope.cc%2Fapp%2Ff3b8cb87d8c932e5673206a3808d227e36964416-1e24dc8c20%3Fdbg%3D1%23editor&data=04%7C01%7Cjhalleux%40microsoft.com%7Ca91fb2906b024bafe0bd08da1322dcab%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637843337705718104%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=p9EeeQZkQQw%2F8Ak7xOkzMfoN075p4RiRJ1ChANN%2Fs8k%3D&reserved=0>

[image]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F3764089%2F161079227-a78fa9e2-2a0d-458c-b3bc-d8c68b82d40d.png&data=04%7C01%7Cjhalleux%40microsoft.com%7Ca91fb2906b024bafe0bd08da1322dcab%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637843337705718104%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=FwkjHWobHIkzcA0sV7AIvrYWKNJaitI62wp7JGp46N4%3D&reserved=0>
[image]<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F3764089%2F161079352-d26b5406-8a8a-4124-b5de-a0dc1fcb9e58.png&data=04%7C01%7Cjhalleux%40microsoft.com%7Ca91fb2906b024bafe0bd08da1322dcab%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637843337705718104%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=LeK19Fqpv4koWkG9gsKGtVP6BoH47CxYomjqeednk5s%3D&reserved=0>

Amerlander, 2022-03-31:
Can you elaborate on this, aren't we already using yotta?

Amerlander, 2022-03-31:
ok, I see we have some settings in [/bluetooth/pxt.json](https://github.com/microsoft/pxt-calliope/blob/ec378c5b09bffd04bfc55fb4198304688c8bb861/libs/bluetooth/pxt.json#L20) and some in [/core/pxt.json](https://github.com/microsoft/pxt-calliope/blob/ec378c5b09bffd04bfc55fb4198304688c8bb861/libs/core/pxt.json#L105), so it might interfere there. Everything is already in yotta.

I`ll test this local and come back with a PR or more questions :) thank your for your help!

Amerlander, 2022-04-02:
I did some more testing:

- Change from 32K to 16K per switch and click `save` does not change the hex
- Changing from 32K to 16K per switch, click `save` and click on the button `edit settings in text mode` changes the hex as expected – even if  I do not change anything in text mode. But I have to download the hex file while the text mode is active, if I switch back to the blocks view before downloading the hex file it does not apply the settings to the hex file.

Amerlander, 2022-05-16:
Having it on https://makecode.calliope.cc/beta it now seems to be a bit better and does not need to recompile every time it switches to 32K. But still sometimes caches the old setting. So switching from 16K to 32K or from 32K back to 16K works in general, but not every time and is a bit laggy and not really predictable.

Amerlander, 2022-08-11:
@pelikhan this issue still persist.

Being in blocks-mode MakeCode always delivers a hex file with the default settings, no matter what setting is selected. Switching to JS mode delivers a hex file with the selected settings.

A hex file for 32K is about 8-10kb bigger than a hex file for the 16K device.

## To reproduce

1. Open https://makecode.calliope.cc/beta
2. Open a new project
3. set a simple program, like showing a Hearth on start.
4. Open project settings and choose "32K RAM"
5. Click on save, editor goes back to blocks
6. Download file, notice, that the file size is now 653.08 KB
7. Switch to JS Mode
8. Download again, notice, that the file size is now 661.48 KB

- Both files should be the same size and run only on a calliope mini with 32K RAM. 
- The first file will run on all Calliope mini devices
- The second file will only run on Calliope mini devices having 32K RAM.

Amerlander, 2022-08-22:
@riknoll could you look into this?

> Being in blocks-mode MakeCode always delivers a hex file with the default settings, no matter what setting is selected. Switching to JS mode delivers a hex file with the selected settings.
> 
> A hex file for 32K is about 8-10kb bigger than a hex file for the 16K device.
> 
> ## To reproduce
> 1. Open https://makecode.calliope.cc/beta
> 2. Open a new project
> 3. set a simple program, like showing a Hearth on start.
> 4. Open project settings and choose "32K RAM"
> 5. Click on save, editor goes back to blocks
> 6. Download file, notice, that the file size is 653.08 KB
> 7. Switch to JS Mode
> 8. Download again, notice, that the file size is now 661.48 KB
> 
> * Both files should be the same size and run only on a calliope mini with 32K RAM.
> * The first file will run on all Calliope mini devices
> * The second file will only run on Calliope mini devices having 32K RAM.

joernalraun, 2022-08-24:
Doing a manual a reload after changing the settings works.
If it would be possible to do an automatic reload after clicking "Save" it might be a quick fix to make it work.

closed 2022-09-17 by Amerlander