microsoft/pxt-calliope #168: Check deprecated Blocks and map them to new blocks if possible

Amerlander, 2022-03-24:
We have some blocks deprecated and should consider to remove them and map their functions to the new blocks if possible:

- [ ] `basic.color()`
- [ ] `basic.rgbw(red: number, green: number, blue: number, white:number)` => `basic.rgb(red: number, green: number, blue: number)`
- [ ] `input.onButtonPressed(Button button, Action body)` => `input.onButtonEvent(Button button, int eventType, Action body)`
- [ ] `input.onPinPressed(TouchPin name, Action body)` => `onPinTouchEvent(TouchPin name, int eventType, Action body)`
- [ ] `input.onPinReleased(TouchPin name, Action body)` => `onPinTouchEvent(TouchPin name, int eventType, Action body)`
- [ ] `input.loudness()` => `input.soundLevel()`

Amerlander, 2022-04-04:
@pelikhan can you give me a hint on this:

I try to map old functions to new ones using the `compile⇒patches` section in `pxtarget.json`:

To map `basic.showArrow(ArrowNames.North)` to `basic.showIcon(IconNames.ArrowNorth)` I used this code:
```json
"0.0.0 - 3.0.33": [
    {
        "type": "api",
        "map": {
            "basic\\s*\\.\\s*showArrow\\s*\\(\\s*ArrowNames\\s*\\.\\s*": "basic.showIcon(IconNames.Arrow"
        }
    }
]
```

But when import a hex file containing the old blocks, they are not mapped to the new blocks.
Is there something I have to know, or did I misunderstand the functionality of the map patches?

pelikhan, 2022-04-04:
@riknoll

riknoll, 2022-04-04:
@Amerlander that's just for updating ts files. To update blocks, you want to update this function:

https://github.com/microsoft/pxt-calliope/blob/master/editor/patch.ts

If you need to see what the old/new blocks XML look like, you can add `?saveblocks=1` to the editor url (before the `#editor`). This will make it so that the save button (the floppy disk, not the download button) will save the blocks file of the project.

Amerlander, 2022-04-22:
@riknoll can you help me one more time?
Updating the blocks worked fine, thank you for that! But the TS patches are somehow buggy and I'm not sure how to debug these rules.

My mapping looks like this, but only the first three rules are working, the three `input...` rules are doesn't change anything:
```json
"type": "api",
"map": {
  "basic\\s*\\.\\s*showArrow\\s*\\(": "basic.showIcon(",
  "images\\s*\\.\\s*arrowImage\\s*\\(": "images.iconImage(",
  "ArrowNames\\s*\\.\\s*": "IconNames.Arrow",
  "input\\s*\\.\\s*onButtonPressed\\s*\\(\\s*(.*?),": "input.onButtonEvent($1, ButtonEvent.Clicked,",
  "input\\s*\\.\\s*onPinPressed\\s*\\(\\s*(.*?),": "input.onPinTouchEvent($1, ButtonEvent.Down,",
  "input\\s*\\.\\s*onPinReleased\\s*\\(\\s*(.*?),": "input.onPinTouchEvent($1, ButtonEvent.Up,"
}
```

I e.g. try to import a hex file containing a main.ts like this:
```ts
input.onPinPressed(TouchPin.P0, function () {
	
})
input.onPinReleased(TouchPin.P2, function () {
	
})
input.onButtonPressed(Button.A, function () {
	
})
input.onButtonPressed(Button.B, function () {
	
})
```
Which just stays the same. The desired output would be:
```ts
input.onPinTouchEvent(TouchPin.P0, ButtonEvent.Down, function () {
	
})
input.onPinTouchEvent(TouchPin.P0, ButtonEvent.Up, function () {
	
})
input.onButtonEvent(Button.A, ButtonEvent.Click, function () {
	
})
input.onButtonEvent(Button.B, ButtonEvent.Click, function () {
	
})
```

riknoll, 2022-04-25:
Hmmm... that is odd! let me take a look

riknoll, 2022-04-25:
@Amerlander did you update the project version that the rules run on? e.g.

```
"0.0.0 - 1.4.12"
```

Also make sure the project you are using for testing was shared from a version of the editor within that range.

Amerlander, 2022-04-25:
I did it like above using `"0.0.0 - 3.0.33"` as well as like in this PR draft `"0.0.0 - 4.0.19"`: https://github.com/microsoft/pxt-calliope/pull/175/files#diff-81e10b85a2c1133c6052389a7a8f019ae31adc4ac985a76887b253d47c2ad202R101

In both cases the first rules are working, therefore I think that this part works.

Amerlander, 2022-04-25:
@riknoll I just made two test files using https://makecode.calliope.cc/#editor (v3.0.33) in JavaScript mode:

First:
```js
basic.showArrow(ArrowNames.East)
```
which I import to the local hosted version of https://github.com/microsoft/pxt-calliope/pull/175/ and it gets successful changed to:
```js
basic.showIcon(IconNames.ArrowEast)
```

Second:
```js
input.onButtonPressed(Button.A, function () {
	
})
basic.showArrow(ArrowNames.East)
```
Which does not change at all when I import it, so even the line `basic.showArrow(ArrowNames.East)` is not changed.

riknoll, 2022-04-25:
For reference, here is the code that it produces:

```ts
input.onButtonEvent(Button.A, ButtonEvent.Clicked, function () {
	
})
basic.showIcon(IconNames.ArrowEast)
```

This does indeed fail to compile if you paste it into the TS editor

Amerlander, 2022-04-26:
Oh, there is a typo on my fault, thought I triple checked this. Sorry, and thank you for your patience!