microsoft/pxt-calliope #170: generated .hex files don´t run on Calliope V1.3

MKleinSB, 2022-03-26:
Generated hexfiles only run on Calliope V2 but not on Calliope V1.3 (V1.2 not tested)
[image]

joernalraun, 2022-03-26:
This is because of the fact that we will be able to make use of the 32KB of the Rev2.0 processor. This files are not compatible! You can change this within the settings of your current project. However, the switch has no effect currently (read more about this at #161).

MKleinSB, 2022-03-26:
I think this way is not easy enough for Children to find. No way to autodetect the version of the mini?

Amerlander, 2022-03-26:
For the final version, we will try to get a better solution – maybe a dialog when you open a new project to choose your device. But first we have to be able to switch the settings at all. Also, the 16KB device has some other problems in the new version and is more likely to crash, so that will also be something we have to solve.

Amerlander, 2022-03-26:
If we get it stable on the 16KB devices, we could also make 16KB the default (which works on both devices) and have the 32KB as an option.

MKleinSB, 2022-03-27:
That would be the most practicable solution I think. For most schoolprojects 16k would be enough if not, you can switch to 32kb.

closed 2022-04-04 by riknoll