microsoft/pxt-calliope #173: Using umlaute in function names crashes the .hex

joernalraun, 2022-04-04:
You can make use of Umlaute (such as ö, ä, ü, or even ß) in function names.
When saving a hex file from this and trying to open it, the browser window becomes unresponsive after a while. All changes (before it doesn't react anymore) are not saved to the file.

It would be good if certain letters could not be used (and/or a message appears when using them) in such instances.

Amerlander, 2022-04-04:
I can not reproduce it.

Can you provide a sample hex file and check if this also appears in Firefox and Chrome (I suppose you used safari)?

Does this also happen in https://makecode.microbit.org/#editor ?

joernalraun, 2022-04-04:
It has been the case on Windows 10 with Edge on several computers and is also the case on makemode.microbit.org (from what has been reported).

Amerlander, 2022-04-04:
Well, can reproduce it now. The error message in Edge claims "Out of Memory".

- Opening https://makecode.microbit.org/_VPoUAzAo1dRa in Edge, Vivaldi or Firefox on Windows 10 ( I didn't test other systems)
- Click on "Edit"
- Notice, that the Simulator is not loading
- Wait, without any interaction:
    - Edge crashes after about 25 seconds
    - Vivaldi crashes after about 35 Seconds
    - Firefox does not seem to crash, but the simulator refuses to work
    
@pelikhan I guess this is something that has to be fixed in https://github.com/microsoft/pxt

Amerlander, 2022-04-04:
A simpler version (less special characters) is loading the Simulator and seem to work: https://makecode.microbit.org/_2HcbJRbcdHya

Amerlander, 2022-04-04:
Its working well in https://arcade.makecode.com/45552-86181-46605-09798, so I guess it is not a pxt thing, but something pxt-calliope and pxt-microbit having in common.

elssner, 2022-04-10:
This arcade example crashes, when you rename the function to Fußgänger-Ampel. Then go to Home, open the saved project again, change from blocks to javascript and back ... the browser will crash with out of memory.

elssner, 2022-04-10:
From the function name 'ß-A' the '-' is removed in javascript to 'ßA'. This '-' can not be restored to blocks when the name includes 'ß' and the project is re-opened. The function with this name must not be called, it must only exist.

elssner, 2022-04-10:
function ßA () {
	
}
game.splash("Start")

[image]

ralf-krause, 2023-01-03:
Is the umlaut problem fixed in the current version of MakeCode?
I tested function names and variable names with German and some French umlauts. I tested the desktop browsers Firefox, Chrome, Chromium, Safari (on macOS) in their current versions. I also got no problems in the current Calliope App on iPadOS and Android.

Here is my test program  https://makecode.calliope.cc/_2F62y97F4Yku

[image]

elssner, 2023-01-07:
In the current version is no umlaut problem, when I write new code. 

When I import the old hex-File with this problem, the Edge Browser crashes again.

[mini-Ampel5.zip](https://github.com/microsoft/pxt-calliope/files/10366547/mini-Ampel5.zip)

ralf-krause, 2023-01-07:
What happens when you import my linked project https://makecode.calliope.cc/_2F62y97F4Yku

ralf-krause, 2023-01-08:
Is it possible to edit the linked test project? 
Can you use the variable to set a new value? 
Can you upload the hex file to the Calliope?
Does the Calliope show this value on the 5x5 display?

elssner, 2023-01-08:
I can edit your project in blocks and in javascript, change the values, upload an see it on the display. It works also whe I rename the function with spaces: usingÜ and ÄinFunctionName

MKleinSB, 2023-01-08:
@elssner makecode.calliope works fine, but you use arcade.makecode.com
so use https://github.com/microsoft/pxt-arcade/issues for issues.