microsoft/pxt-calliope #178: Tutorials stopped working in v3.0.33

Amerlander, 2022-04-11:
Tutorials stopped working in the stable Version(v3.0.33). 

They either do not appear at all, or they are shown, but can't be opened.

**To Reproduce**
Steps to reproduce the behavior:
1. Clear cache or use an incognito browser session
2.  Go to https://makecode.calliope.cc
3a. No tutorials or links are shown
3b. The tutorials are showing up, but they can not be opened.

**Expected behavior**
Tutorials should be visible and open.

**Screenshots**
[image]

[image]

Amerlander, 2022-04-11:
@riknoll this is affecting the stable version. Can this be caused by our latest updates?

riknoll, 2022-04-11:
Yes, this was a result of changing the major version to 4. I created a v3 branch in the repo that points to the pre-v4 changes and this should be fixed now

Amerlander, 2022-04-11:
yes, seems fixed now.

closed 2022-04-11 by Amerlander