microsoft/pxt-calliope #179: .blocks tests

Amerlander, 2022-04-25:
I updated the .blocks files like described there: https://makecode.com/develop/blockstests

That resulted in new errors, telling there are XML nodes missing. I had to remove these XML Nodes to pass the tests: https://github.com/microsoft/pxt-calliope/pull/175/commits/9534c8165d70f8ed7e17322cf6d0deda3a9b3307

I thought actually the tests should never fail, when the .blocks files are directly saved from the editor. Are these tests outdated, or did I something wrong? @riknoll, can you help?

riknoll, 2022-04-25:
hmmmm this could be a mismatch in the version used for testing and the version sued to generate the blocks file. Did you do both of those things on localhost?

Amerlander, 2022-04-25:
Yes, both in localhost and on the same machine