microsoft/pxt-calliope #214: Missing translations

Amerlander, 2023-01-16:
There are some parts of MakeCode that are not translated to german:

- [ ] "remainder of %1 / %2" is not translated. Should be "Rest von %1 ÷ %2"
- [ ] "serial write buffer" is not translated. Should be "seriell|schreibe Zwischenspeicher %buffer"
- [ ] Share Popup: Info-Box and Button

[image]
[image]

joernalraun, 2023-01-16:
One thing to add: this items can‘t be translated through Crowdin, though they are there (and translated), but don‘t show up on MakeCode.

joernalraun, 2023-02-16:
Are there any news on this – where should it be posted instead? This is a problem throughout all MakeCode platforms.

MKleinSB, 2023-02-16:
[image]

Serial write buffer is fixed now. 

described in
https://github.com/microsoft/pxt-microbit/issues/4873

[image]
 
remainder has no String in Crowdin. I can´t fix it in  In-Context translation. The string disappeared around a year.

joernalraun, 2023-02-16:
I still see Serial write buffer, but maybe it has just been fixed and it is still cached somewhere.
Remainder is also still not translated. The string seems to be available in Crowdin, but also other languages have this problem with it.

MKleinSB, 2023-02-16:
Yes, it takes a day or so. 
makecode.calliope.cc/?translate=1#editor

MKleinSB, 2023-02-16:
https://github.com/microsoft/pxt-microbit/issues/4904
 I think Galen Nickel is the right one but don´t want to post it twice

MKleinSB, 2023-02-23:
Hm, after week the translated and approved text still does not appear in makecode. It appears only in in-context-translation
[image]

MKleinSB, 2023-03-14:
Now serial writebuffer now really ist fixed! 
remainder still remains :-(

closed 2023-11-08 by Amerlander