microsoft/pxt-calliope #216: Controlling a motor with different speeds | Help wanted

RaghuvirShirodkar, 2023-01-25:
Hello there!

I am having problems controlling a motor with different speeds. Would it be possible to look at the implementation file for the following function? I can't seem to find it:
https://github.com/microsoft/pxt-calliope/blob/b0af4b54b2a8418fd7c9b364ece3fe2aeb70d5c1/libs/core/motors.cpp#L33

Any help would be greatly appreciated!

Best,
Raghuvir

joernalraun, 2023-01-25:
Dear Raghuvir,

could you specify the problem a bit more?
Maybe you are referring to this:
Using the motor blocks need usually a higher input value until the motor starts, because it needs some voltage to start turning.
However, maybe we can change the ratio of the input value so it starts turning with smaller values? If this is done it should also be communicated to Open Roberta, so both editors are behaving in the same way...

Best
Joern

RaghuvirShirodkar, 2023-01-25:
Hello Joern,

Thank you for your prompt reply!

The problem generally occurs when controlling a single motor with lower speeds in both directions. Instead of turning slower, the motor sometimes seems to continue at full power. It works, however, when +-100% is provided as an input. 

Maybe the implementation of the aforementioned function (`uBit.soundmotor.motorOn()`) could shed light on the inner workings. Would you happen to know where I can find it?

Best,
Raghuvir

joernalraun, 2023-01-25:
Ah, ok. So far I wasn't able to see such a behavior.
I think what you are looking for should be rather in the DAL, somewhere here:
https://github.com/calliope-mini/microbit-dal/blob/master/source/drivers/CalliopeSoundMotor.cpp

Best
Joern

Amerlander, 2023-11-17:
I'll close this here. Feel free to reopen and provide additional info if this still need a fix in makecode.

closed 2023-11-17 by Amerlander