microsoft/pxt-calliope #238: No WebUSB-pairing when V1 or V2 is chosen

MKleinSB, 2023-10-24:
There is no WebUSB-functionality when calliope V1 or V2 is chosen in makecode **V6.0.22**. When V3 is chosenit works perfectly with V1, V2 and V3. 

[image]

joernalraun, 2023-10-26:
Right now, there are some issues with WebUSB for older devices. We are trying to fix this and open it for all other hardware as well.

Amerlander, 2023-11-14:
I created seperate issues for webUSB:

For Calliope mini 2: https://github.com/microsoft/pxt-calliope/issues/248

For enabling support for Calliope mini 1 while mini 2 is not suporting it: https://github.com/microsoft/pxt-calliope/issues/250

WebUSB will be disabled for mini 1 and 2 in the next version until these issues are not closed. But it will be possible to enable webUSB for all devices by adding `?usb` to the url. Still makecode wont be able to connect to the mini 2.

closed 2023-11-14 by Amerlander