microsoft/pxt-calliope #250: Enable/Disable WebUSB by board instead of compile service

Amerlander, 2023-11-08:
Currently the availablility of webUSB is defined in `pxtarget.json`.`compile` and could be overwritten by defining different compile variants.

We have three baord definitions, two of them are sharing one compile variant. Since one board do not allow webUSB yet it has to be disabled for both. Is there a way to enable/disable webUSB in pxt.json of the board core library instead of the compile service?

Current webUSB flag:
https://github.com/microsoft/pxt-calliope/blob/4cc77034fa7b8a88818e2d91afd7aa63926dd83e/pxtarget.json#L159

mini 1 core package (should support webUSB)
https://github.com/microsoft/pxt-calliope/blob/master/libs/v1/pxt.json

mini 2 core package (should not yet support webUSB)
https://github.com/microsoft/pxt-calliope/blob/master/libs/v2/pxt.json

@jwunderl can you tell if this is possible somehow? If https://github.com/microsoft/pxt-calliope/issues/248 is solved we would not require to disable webUSB for a specific device, but in the meanwhile it would be good to be able to enable web usb for the Calliope mini 1 devices.

Amerlander, 2023-11-17:
@abchatra can we get help on this?

abchatra, 2023-11-21:
@riknoll any workarounds here?

riknoll, 2023-11-21:
this is not supported today AFAIK and would not be trivial to add... @Amerlander couldn't you just duplicate the compile variant and change that one setting? I'm not sure I understand why they have to share it.

Amerlander, 2023-11-22:
Thank you for that feedback. Dublicating the compile variant is an option, I'll do that.

closed 2023-11-22 by Amerlander