microsoft/pxt-calliope #255: First random number always the same

jasperp92, 2023-11-15:
The block "random int" generates the same pseudo random number after reset. It looks like it shows the seed, but has not generated a random number yet.

kreutel, 2023-11-16:
Fount that one too. The RNG gets seeded identically every time - or not at all. Anyway: The sequence of random numbers is always the same.

Amerlander, 2023-11-16:
I'm working on it. Choosing the calliope mini 1 template should still give random numbers.

Amerlander, 2023-11-17:
> I'm working on it. Choosing the calliope mini 1 template should still give random numbers.

Sorry, I was talking about the beta version.

Seems related to the bluetooth stack. I Can reproduce it on the microbit when I load ble: https://makecode.microbit.org/S81402-73332-20016-17530
In the beta version bluetooth is only loaded for mini 2 and mini 3 devices, that's why the mini 1 template shows random numbers.

The random seed generation falls back to a static seed if ble is enabled: https://github.com/calliope-edu/codal-microbit-v2/blob/dc0383b581156ef8196202b299a629c6c29d6ba9/source/MicroBitDevice.cpp#L67

Usually, the `ubit.init()` function should generate the seed before ble actually starts, so that *should* not be an issue.
I also was not able to reproduce this issue using the samples with enabled ble.
I also tried to generate a random seed by sensor data, but this has no effect: https://github.com/calliope-edu/codal-microbit-v2/blob/beed355baa7308c176aeef4ea5e316ddcc74f7a8/model/MicroBit.cpp#L215

closed 2025-03-13 by Amerlander