microsoft/pxt-calliope #269: Importing an old Hex into a new project switches hardware

joernalraun, 2023-11-29:
When trying to import a V1/V2 Hex file into a V3 project the hardware switches to V1/V2 and the project settings switch to V1 (no Bluetooth activated). The switch can't be turned to on again, even when manually switching back to V3 hardware.
This means old projects currently can't be used mobile on V3!

joernalraun, 2023-11-29:
Select Calliope V3 as hardware and use
 this file:
[mini-Test-18.hex.zip](https://github.com/microsoft/pxt-calliope/files/13501215/mini-Test-18.hex.zip)
(from Calliope mini V2)
The hardware than switches and you will not be able to change the bluetooth settings (first option) by hand.
The option niche to change the settings might be correct as radio is imported, but the change of the hardware seems not correct...

Amerlander, 2023-11-29:
Its the `Environment_IoT` Extension, that requires the `radio` extension. Radio disables Bluetooth.

[image]

Amerlander, 2023-11-29:
If you import a program that was made in MakeCode 4 or before you will always have to select the hardware **after** the import. It does not matter what hardware was selected first or how you import the old Hex file (shared link, imported hex file in a project, ...)

joernalraun, 2023-11-29:
Ah, I thought there is a different way to swap versions. However, we will nw communicate to first load the file and than switch the version!

closed 2023-11-29 by joernalraun