microsoft/pxt-calliope #272: Analog set period sets the same frequency on all pins, even if API call specifies pin

fabianhugo, 2023-12-20:
As in pxt-microbit issue [4950](https://github.com/microsoft/pxt-microbit/issues/4950) the PWM frequency set by the "**analog set period pin P... to (us) 20000**" sets the PWM frequency globally (default is 50Hz). 
Any follow up on this issue? Can this be fixed?

Amerlander, 2023-12-20:
Might be a codal issue, the function is defined here: https://github.com/lancaster-university/codal-nrf52/blob/0643733703b4f8f788af81a996f40d1f1a1527bc/source/NRF52Pin.cpp#L643

fabianhugo, 2023-12-20:
The behavior of a program on the base of micro:bit samples v2 is the same. Example code:     uBit.init();
```
    while (1){
    uBit.sleep(3000);
    uBit.io.P0.setAnalogPeriodUs(100);
    uBit.io.P0.setAnalogValue(10);
    uBit.sleep(3000);
    uBit.io.P1.setAnalogPeriodUs(500);
    uBit.io.P1.setAnalogValue(10);
    }
```

Amerlander, 2024-01-25:
The block is defined in pxt-common packages and designed for boards that allow separate PWM frequencys.
The CODAL does not support this, so I'll close it here.

If this feature is required, it has to be changed in the CODAL.

closed 2024-01-25 by Amerlander