microsoft/pxt-calliope #275: SerialNumber doesnt show up

jasperp92, 2024-01-04:
Noticed a small bug:

If I use the nlock SerialNumber I cant read the serial data in the serial monitor:

basic.forever(function () {
    serial.writeNumber(input.acceleration(Dimension.X))
})

The Show data "Device" isnt't popping up and I get no values.

Serial Line works fine tho

Amerlander, 2024-01-25:
I can not reproduce this, although I would always prefer a new line at some point.

[image]

Amerlander, 2024-01-26:
I missunderstood.

Its a pxt-core issue. MakeCode is receiving data, but the simulator only processes the data when it receives a lineending:

```
input.onButtonPressed(Button.A, function () {
    serial.writeLine("")
})
basic.forever(function () {
    serial.writeNumber(input.acceleration(Dimension.X))
})
```

With this code, the data from `writeNumber` is displayed everytime the empty Line is written (Button A).
I think this is a design limitation, and it won't be fixed in pxt-core. if this still needs a fix or workaround, the issue should be reopened in pxt-core.

closed 2024-01-26 by Amerlander