microsoft/pxt-calliope #296: Simulator not triggering events in latest chromium based browsers

Amerlander, 2024-07-02:
**Description:**
After the update to Microsoft Edge Version 125.0.2535.79 and Google Chrome Version 125.0.6422.142, the simulator in MakeCode no longer triggers events when buttons or pins are interacted with. The inputs seem clickable, but does not react visually on click. This issue does not occur in other browsers or older versions of Edge and Chrome.

**Steps to Reproduce:**
1. Open MakeCode on a Windows 10 or 11 machine with Edge Version 125.0.2535.79 or Chrome Version 125.0.6422.142.
2. Create a simple program with events such as "when button A is pressed" or "when pin 0 is touched".
3. Run the simulator and try interacting with the buttons or pins.

**Expected Behavior:**
The simulator should react on interaction and execute the code within the event blocks when buttons or pins are interacted with.

**Actual Behavior:**
The simulator does not react on clicks but does not execute the code within the event blocks.

**Additional Information:**
- The issue is present with the latest MakeCode version and also with version 4.
- The issue does not occur in Safari on iPad, in the Edge app with version 125.0.2535.72, or on Windows 8 with the old Edge.
- makecode.microbit.org seems to work correctly in the current Edge version.
- The issue was confirmed by other users who also reported that no button events are triggered and no analog values can be input at the pins.
- The problem is not reproducible when the editor is started with `pxt serve` or hosted locally with `statickpkg`.
- There are no errors in the JavaScript console, making it difficult to debug the issue without guessing.

**Forum Discussion (german):**
- [Edge Version 125.0.2535.79 = Simulator kaputt?](https://forum.calliope.cc/t/edge-version-125-0-2535-79-simulator-kaputt/2903/4)

**System Information:**
- **Browser:** 
  - Microsoft Edge Version 125.0.2535.79
  - Google Chrome Version 125.0.6422.142
  - Other Chromium based browsers like Opera and Vivaldi on latest Versions
- **Operating System:** Windows 10 and Windows 11
- **MakeCode Version:** Latest and v4

**Steps to Fix or Investigate:**
1. Verify if the issue can be reproduced on different systems with the same Edge and Chrome versions.
2. Investigate potential changes in Edge and Chrome updates that might be causing the issue.
3. Compare the behavior of MakeCode on the affected browser versions with other browsers to identify potential compatibility issues.
4. Since the issue does not appear when hosted locally, consider differences in the hosting environment or configuration that might affect simulator behavior.

**Notes:**
- the issue resolves when a sensor is read, for example, writing light intensity to a variable or displaying it on the screen.

Please look into this issue as it significantly impacts the usability of the simulator for users on the latest versions of Edge and Chrome. Thank you!

Amerlander, 2024-07-05:
@abchatra I'm stuck on this issue. I don't really have any idea how to debug it. Do you know why the simulator behaves differently on the hosted version compared to my development environment?

abchatra, 2024-07-05:
@riknoll ?

riknoll, 2024-07-08:
will take a look today!

riknoll, 2024-07-08:
I'm able to repro this locally, seeing the same behavior as @Amerlander where it doesn't repro when serving from the cli. I also have confirmed that this isn't a case of one event listener preventing propagation to another; it seems like the browser is genuinely not firing the events on the hosted site.

Kind of stumped as to what to do here, but at least I can reproduce it easily. I'll try a few things and see if any of them happen to fix things. Will update when I have more info.

closed 2024-07-10 by riknoll