microsoft/pxt-calliope #316: Incorrect transmission of character strings via serial interface

maomen1975, 2025-04-07:
**Describe the bug**
I create the following program in makecode.calliope.cc: 

input.onButtonEvent(Button.A, input.buttonEventClick(), function () {
   basic.pause(100)
   serial.writeLine("Hello World!")
   basic.pause(100)
})
serial.redirectToUSB()
serial.setBaudRate(BaudRate.BaudRate115200)

Regardless of which program I use on the PC to receive the data from the Calliope (e.g. TeraTerm, Arduino IDE, makecode data plotter), I receive the following data in these programs when sending multiple times:

aloWlt
Halo Welt!
HalloWlt
allo Wel!
Hal Welt!

When I program the Calliope with Python or OpenRoberta, the string is received correctly: Hallo Welt!

**To Reproduce**
Steps to reproduce the behavior:

See above.

**Expected behavior**

I expect to receive the string "Hallo Welt!"

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser: Google Chrome (135.0.7049.41), Arduino IDE (2.3.5), Tera Term (5.3)

joernalraun, 2025-04-08:
When running the program inside of the MakeCode monitor, it works as expected.

maomen1975, 2025-04-09:
Hello, I have found a mistake. A stupid error. The makecode app was still connected to the Calliope. Therefore TeraTerm and ArduinoIDE only partially received the string sent by the Calliope. Since something was received at all, I didn't realize that it was because the makecode/Calliope connection was still active. However, everything works perfectly now

closed 2025-04-09 by maomen1975