microsoft/pxt-calliope #317: Extension search returns 500 error

Amerlander, 2025-04-10:
### **Bug description**

When trying to search for extensions, no results are returned.

The browser console shows a `500` error for requests like: 
[https://makecode.calliope.cc/api/ghsearch/calliopemini/calliopemini?q=grove](https://makecode.calliope.cc/api/ghsearch/calliopemini/calliopemini?q=grove)

### **To Reproduce**
Steps to reproduce the issue:
1. Go to [makecode.calliope.cc#editor](https://makecode.calliope.cc#editor)
2. Click **Add Extension**
3. Search for an extension (e.g. “Grove”)
4. No extensions are found. A 500 error appears in the console.

### **Expected Behavior**
Searching for extensions should return matching results.

### **Screenshots**

[image]

abchatra, 2025-04-10:
@eanders-ms ?

eanders-ms, 2025-04-10:
Thanks for reporting, and apologies for this. We have made a fix server-side and although I can't give a firm timeline, we will get it deployed as soon as possible.

eanders-ms, 2025-04-10:
Extension search should be working again. The issue has to do with server-side access to redirected GitHub repos. We'll roll out a that fix for that separately. In the meantime, we've updated your targetconfig.json to not include the old repo reference (new one was already present).

old: microsoft/pxt-hacking-stem
new: microsoft/pxt-data-streamer

closed 2025-04-10 by eanders-ms